AdyRock/ady.sofar #1: supported Solarman devices

JensDesm, 2024-03-06:
It's not clear which devices are supported.
Are all (4G/GPRS/WiFi/Ethernet) Solarman Stick loggers supported or only the ethernet version?
We have the LSW-3 2.4G WiFi. Will it work?

Thanks in advance for update.

AdyRock, 2024-03-06:
Wi-Fi or Ethernet are fine, provide the inverter is supported. The app requires a local connection, so the 4G/GPRS probably won't work.

Rrrr4321, 2024-04-01:
Hi Adrian,
Is there a way you can show 2 decimals on Total energy?

https://community.homey.app/t/app-obsolete-sofar-solarman/70830/256

AdyRock, 2024-04-01:
Thanks for the nudge as I seemed to have missed that one. I have updated that topic now with the following:

Sorry, I seemed to have missed this one.
Do you mean Total energy in the Solar panel device?
Or would that be Today in the Solar panel device?

Depending on the inverter model, the total generation is reported to either 1 decimal place or 0 decimal places. The capability is a standard 'meter_power' which is set to 2 decimals.

So, as far as I can see, there is no rounding that I have any control over.

FGFME, 2024-04-26:
Bonjour,
Je suis équipé d'une solution sofar Solar (onduleur, batterie, transmetteur wifi)
J'ai installé l'App dans HOMEY 
Cependant aucune donnée remontée dans HOMEY
Mon modèle d'onduleur est HYD-30KTL-3PH
Je suis sur l'application SOFARMAN Business pour avoir les stats en temps réel mais je souhaite avoir ces infos sous HOMEY pour activer des scripts 
Ai je raté quelque chose ?
Merci pour votre soutien 
A bientôt

AdyRock, 2024-04-26:
Could you enable the Diagnostic Log, restart the app and then send the log.
[image]

FGFME, 2024-04-26:
Voici le lien pour le diagnostic 
052c184f-2fc5-438d-a93e-4c7cc86920de

Et les logs générés suite au redémarrage de l'appli
* 2024-04-26T07:08:49.484Z
************** App has initialised. ***************

* 2024-04-26T07:08:49.694Z
Get Data

* 2024-04-26T07:08:54.641Z
Found Inverter: : 192.168.21.12, 2331343663

* 2024-04-26T07:08:54.643Z
Checking register 14 for grid frequency:

* 2024-04-26T07:08:54.763Z
Returned null.

Checking register 1156 for grid frequency:

* 2024-04-26T07:08:54.851Z
Frequency 50.03 is good

* 2024-04-26T07:08:54.852Z
Found inverter

* 2024-04-26T07:08:59.696Z
Get Data

* 2024-04-26T07:09:00.381Z
No Data

* 2024-04-26T07:09:10.384Z
Get Data

* 2024-04-26T07:09:12.084Z
No Data

* 2024-04-26T07:09:22.119Z
Get Data

* 2024-04-26T07:09:22.868Z
No Data

* 2024-04-26T07:09:32.876Z
Get Data

* 2024-04-26T07:09:33.569Z
No Data

* 2024-04-26T07:09:43.579Z
Get Data

* 2024-04-26T07:09:44.283Z
No Data

* 2024-04-26T07:09:54.286Z
Get Data

* 2024-04-26T07:09:54.944Z
No Data

FGFME, 2024-04-26:
l'IP indiquée dans le log correspond bien a l'onduleur

FGFME, 2024-04-26:
Je pensais désinstaller et réinstaller l'application. Bonne ou mauvaise idée ?

FGFME, 2024-04-26:
Je vais peut être redémarrer l'onduleur car depuis la réinstallation de l'application j'ai une erreur http dans lesnlog de l'application sofar

AdyRock, 2024-04-26:
I don't think reinstalling the app will help.
As far as I can see, the HYD-30KTL uses the same registers as the HYD-xxG3 and that is what is being found with the grid frequency in register 1156. I don't know why it then doesn't find anything when polling for the other data.

AdyRock, 2024-04-26:
Maybe it doesn't support all the registers in that profile which is causing an error. Would you be OK to do some testing of individual registers that might take a lot of your time?

FGFME, 2024-04-30:
> solis_hybrid.json
> If you want to, the attached file contains the registers for the G3
> profile. To check it, you need to open the app settings and go to the
> Register Log. Then enter each register number shown in the file,
> one-by-one, to see if it returns data.
> image.png (view on web)
>
> If it fails. maybe try a second time just to make sure it wasn't a
> different error.
> If it always fails, then note it as a bad register and when you have
> finished, send the list of bad registers. I will then crate a new profile
> without those register in.

Bonjour Adrian,
Je serai proche de l'onduleur a partir du 12 mai. Je vais essayer de tester
chaque registre comme indiqué a partir de cette date.
Je vous tiens au courant
A bientôt

AdyRock, 2024-11-30:
I'm not sure. The app does have a map for a Deye inverter, but I don't know if it will match yours. 
All I can suggest is to try.

FrankR255, 2024-11-30:
No it can't find an Inverter

AdyRock, 2024-11-30:
Can you open the app settings page and select the Diagnostics Log tab and turn on Log Data. Then, restart the app and wait for a few minutes before going back to the log and tap on Send Log. 
The app should pick up an mDNS broadcast and then try reading certain registers to find a valid mains frequency to determine which map to use. 

Also, could you let me know the exact model, and I will search for the register definitions.

FrankR255, 2024-11-30:
I have an 
Deye 12KW Hybrid PV- Wechselrichter 3 Physik SUN 12K-SG04LP3-E
Logs Send

FrankR255, 2024-11-30:
SUN-12K-S DEYE 12KW Hybrid PV-Wechselrichter
3Phasig SUN-12K-SG04LP3-E
DEYE 12KW Hybrid Wechselrichter 3Phasig SUN-10K-SG04LP3-EU

FrankR255, 2024-11-30:
THX for your Help...

FrankR255, 2024-11-30:
Yes its the same Network.
There is an Logger:
https://de.deyeinverter.com/product/accessory-monitoring-1/stick-logger.html

FrankR255, 2024-12-04:
Now your App found my Inverter.
After a reinstall.
Logs ar send to you.
Maybe you have a look on it

AdyRock, 2024-12-04:
Ok, so looking a bit more promising, but it can't find a suitable register map, so I will see if I can find some information.

FrankR255, 2024-12-04:
OK THx al lot

FrankR255, 2024-12-04:
Here the register i guess.

registers = (
	[  62,	'',								'Year / Month',					'u16', 0, 1.00,		'',		0, 0 ],
	[  63,	'',								'Day / Hour',					'u16', 0, 1.00,		'',		0, 0 ],
	[  64,	'',								'Minute / Second',				'u16', 0, 1.00,		'',		0, 0 ],
	[ 500,	'',								'Run state',					'u16', 0, 1.00,		'',		0, 0 ],
	[ 514,	'battery_charge_today_E',		'Battery charge today',			'u16', 0, 0.10, 	'kWh',	1, 0 ],
	[ 515,	'battery_discharge_today_E',	'Battery discharge today',		'u16', 0, 0.10, 	'kWh',	1, 0 ],
	[ 516,	'battery_charge_total_E',		'Battery charge total l',		'u32', 0, 0.10, 	'kWh',	1, 0 ],
	[ 517,	516,							'Battery charge total h',		'u16', 0, 6553.6,	'kWh',	0, 0 ],
	[ 518,	'battery_discharge_total_E',	'Battery discharge total l',	'u32', 0, 0.10,		'kWh',	1, 0 ],
	[ 519,	518,							'Battery discharge total h',	'u16', 0, 6553.6,	'kWh',	0, 0 ],
	[ 520,	'grid_in_today_E',				'Grid buy energy today',		'u16', 0, 0.10,		'kWh',	1, 0 ],
	[ 521,	'grid_out_today_E',				'Grid sell energy today',		'u16', 0, 0.10,		'kWh',	1, 0 ],
	[ 522,	'grid_in_total_E',				'Grid buy energy total l',		'u32', 0, 0.10,		'kWh',	1, 0 ],
	[ 523,	522,							'Grid buy energy total h',		'u16', 0, 6553.6,	'kWh',	0, 0 ],
	[ 524,	'grid_out_total_E',				'Grid sell energy total l',		'u32', 0, 0.10,		'kWh',	1, 0 ],
	[ 525,	524,							'Grid sell energy total h',		'u16', 0, 6553.6,	'kWh',	0, 0 ],
	[ 526,	'load_today_E',					'Load energy today',			'u16', 0, 0.10,		'kWh',	1, 0 ],
	[ 527,	'load_total_E',					'Load energy total l',			'u32', 0, 0.10,		'kWh',	1, 0 ],
	[ 528,	527,							'Load energy total h',			'u16', 0, 6553.6,	'kWh',	0, 0 ],
	[ 529,	'pv_today_E',					'PV energy today',				'u16', 0, 0.10,		'kWh',	1, 0 ],
	[ 534,	'pv_total_E',					'PV energy total l',			'u32', 0, 0.10,		'kWh',	1, 0 ],
	[ 535,	534,							'PV energy total h',			'u16', 0, 6553.6,	'kWh',	0, 0 ],
	[ 536,	'gen_today_E',					'Generator energy today',		'u16', 0, 0.10,		'kWh',	1, 0 ],
	[ 537,	'gen_total_E',					'Generator energy total l',		'u32', 0, 0.10,		'kWh',	1, 0 ],
	[ 538,	537,							'Generator energy total h',		'u16', 0, 0.10,		'kWh',	0, 0 ],
	[ 540,	'DC_transformer_T',				'DC Transformer temperature',	'u16', -1000, 0.10,	'°C',	1, 0 ],
	[ 541,	'heatsink_T',					'Heat sink temperature',		'u16', -1000, 0.10,	'°C',	1, 0 ],
	[ 551,	1000,							'Inverter power status on/off',	'bit', 0, 1,		'',		2, 0 ],
	[ 552,	1000,							'AC relay status',				'bit', 0, 2,		'',		2, 0 ],
	[ 586,	'battery_T',					'Battery temperature',			'u16', -1000, 0.10,	'°C',	1, 0 ],
	[ 587,	'battery_V',					'Battery Voltage',				'u16', 0, 0.01,		'V',	1, 0 ],
	[ 588,	'battery_SOC',					'Battery state of charge',		'u16', 0, 1.00,		'%',	1, 0 ],
	[ 590,	'battery_P',					'Battery power',				's16', 0, 1.00,		'W',	1, 0 ],
	[ 591,	'battery_I',					'Battery current',				's16', 0, 0.01,		'W',	1, 0 ],
	[ 592,	'',								'? Corrected AH',				'u16', 0, 1.00,		'AH',	0, 0 ],
	[ 598,	'',								'Grid phase 1 voltage',			'u16', 0, 0.10,		'V',	0, 0 ],
	[ 599,	'',								'Grid phase 2 voltage',			'u16', 0, 0.10,		'V',	0, 0 ],
	[ 600,	'',								'Grid phase 3 voltage',			'u16', 0, 0.10,		'V',	0, 0 ],
	[ 604,	'',								'Inner grid phase 1 power',		's16', 0, 1.00,		'W',	0, 0 ],
	[ 605,	'',								'Inner grid phase 2 power',		's16', 0, 1.00,		'W',	0, 0 ],
	[ 606,	'',								'Inner grid phase 3 power',		's16', 0, 1.00,		'W',	0, 0 ],
	[ 607,	'',								'Inner grid total power',		's16', 0, 1.00,		'W',	0, 0 ],
	[ 609,	'grid_frq',						'Grid frequency',				'u16', 0, 0.01,		'Hz',	1, 0 ],
	[ 610,	'',								'Inner grid phase 1 current',	's16', 0, 0.01,		'A',	0, 0 ],
	[ 611,	'',								'Inner grid phase 2 current',	's16', 0, 0.01,		'A',	0, 0 ],
	[ 612,	'',								'Inner grid phase 3 current',	's16', 0, 0.01,		'A',	0, 0 ],
	[ 613,	'',								'Out of grid phase 1 current',	's16', 0, 0.01,		'A',	0, 0 ],
	[ 614,	'',								'Out of grid phase 2 current',	's16', 0, 0.01,		'A',	0, 0 ],
	[ 615,	'',								'Out of grid phase 3 current',	's16', 0, 0.01,		'A',	0, 0 ],
	[ 616,	'',								'Grid phase 1 power',			's16', 0, 1.00,		'W',	0, 0 ],
	[ 617,	'',								'Grid phase 2 power',			's16', 0, 1.00,		'W',	0, 0 ],
	[ 618,	'',								'Grid phase 3 power',			's16', 0, 1.00,		'W',	0, 0 ],
	[ 619,	'grid_P',						'Grid total power',				's16', 0, 1.00,		'W',	1, 0 ],
	[ 627,	'inverter_p1_V',				'Inverter phase 1 voltage',		'u16', 0, 0.10,		'V',	1, 0 ],
	[ 628,	'inverter_p2_V',				'Inverter phase 2 voltage',		'u16', 0, 0.10,		'V',	1, 0 ],
	[ 629,	'inverter_p3_V',				'Inverter phase 3 voltage',		'u16', 0, 0.10,		'V',	1, 0 ],
	[ 630,	'',								'Inverter phase 1 current',		's16', 0, 0.01,		'A',	0, 0 ],
	[ 631,	'',								'Inverter phase 2 current',		's16', 0, 0.01,		'A',	0, 0 ],
	[ 632,	'',								'Inverter phase 3 current',		's16', 0, 0.01,		'A',	0, 0 ],
	[ 633,	'inverter_p1_P',				'Inverter phase 1 power',		's16', 0, 1.00,		'W',	1, 0 ],
	[ 634,	'inverter_p2_P',				'Inverter phase 2 power',		's16', 0, 1.00,		'W',	1, 0 ],
	[ 635,	'inverter_p3_P',				'Inverter phase 3 power',		's16', 0, 1.00,		'W',	1, 0 ],
	[ 636,	'inverter_P',					'Inverter total power',			's16', 0, 1.00,		'W',	1, 0 ],
	[ 637,	'',								'Inverter total apparent power','s16', 0, 1.00,		'W',	0, 0 ],
	[ 638,	'inverter_frq',					'Inverter frequency',			'u16', 0, 0.01,		'Hz',	1, 0 ],
	[ 640,	'',								'Backup phase 1 power',			's16', 0, 1.00,		'W',	0, 0 ],
	[ 641,	'',								'Backup phase 2 power',			's16', 0, 1.00,		'W',	0, 0 ],
[... 3 lines cut ...]
[... cut ...]
	[ 672,	'carport_P',					'Carport power',				's16', 0, 1.00,		'W',	1, 0 ],
	[ 673,	'roof_P',						'Roof power',					's16', 0, 1.00,		'W',	1, 0 ],
	[ 676,	'carport_V',					'Carport voltage',				'u16', 0, 0.10,		'V',	1, 0 ],
	[ 677,	'carport_I',					'Carport current',				's16', 0, 0.10,		'A',	1, 0 ],
	[ 678,	'roof_V',						'Roof voltage',					'u16', 0, 0.10,		'V',	1, 0 ],
	[ 679,	'roof_I',						'Roof current',					's16', 0, 0.10,		'A',	1, 0 ],
	[1000,	'status',						'Status',						'bit',  0, 1.00,	'',		1, 0 ]
)

ref_registers = {}

AdyRock, 2024-12-16:
I have hopefully added support for your inverter. I have no way to test it so it might not show the correct values as I might have got the wrong registers or got the factors wrong.
However please let me know how it goes,

FrankR255, 2024-12-17:
Ich kann nicht sagen warum aber die App findet den Logger nicht mehr. Habe schon mehrfach neu gestartet. Hasts du eine Idee?

AdyRock, 2025-03-27:
Oops, very embarrassing to have left this so long, sorry for the delay. Did the invert start reporting again?
If not and you haven't totally lost confidence in me, could you send the log from the app settings page.

FrankR255, 2025-03-27:
Kein Problem
Habe mittlerweile eine andere App gefunden die meine Bedürfnisse voll erfüllt. Danke aber für deine Rückmeldung.

FrankR255, 2025-10-29:
Hi Adrian, 
would it be possible for us to check again if we can get my Deye inverter working?

Regards
Frank