AdyRock/ady.sofar #3: Sofar Battery return low battery warning

blexter, 2024-09-04:
Since a while my battery/Sofar app return battery status 0% to Homey which result in "Low battery alarm", after seconds/minutes the alarm I cleared.
I tried to remove the battery and re-add it - same problem.
Downloaded latest test-version of Sofar-app for Homey - same problem

Best regards

AdyRock, 2024-09-04:
I'm guessing that is a failed read.
I will look into it to make sure capabilities are not updated if something failed.

AdyRock, 2025-03-27:
Can you check the latest test version. I have added some extra checks to validate the data before setting the capabilities.

blexter, 2025-03-28:
Since I while I ran with test version but my devices havn't returned any values. Sent a log through the app yesterday (I think) to you.
Made some troubleshooting today and found out that the device returns data if I downgrade to stable release (1.1.24)

blexter, 2025-03-28:
* 2025-03-28T20:43:43.311Z Found Inverter: IP: 192.168.XX.XX, S.No: 2706289438 * 2025-03-28T20:43:43.313Z Checking register 14 for grid frequency: * 2025-03-28T20:43:43.423Z Returned null. Checking register 1156 for grid frequency: * 2025-03-28T20:43:43.525Z Frequency 50.03 is good * 2025-03-28T20:43:43.526Z Found inverter * 2025-03-28T20:43:58.477Z Missing one or more of Frequency = 50.01, Consumption = 0, Grid_Voltage = 230.60000000000002, Total_Import = 3196.3 

Can also see that (and seen a while) the values (with 1.1.24) stopped update 8h ago.

AdyRock, 2025-03-28:
Ok, so there are no consumption values. Is that to be expected as it seems strange that your home is not using any power at all?

blexter, 2025-03-28:
I have consumption, my Easee equlizer (power meter connected to p1 port on the meter:

[image]

AdyRock, 2025-03-28:
So, does your Sofar inverter not provide consumption data?

blexter, 2025-03-28:
I guess it does.

[image]

AdyRock, 2025-03-28:
Have you added the Summary / Inverter device from the Sofa app?
That should show the home consumption from the inverter. 
What inverter do you have?

blexter, 2025-03-28:
This one? Not updated last 9-11h.
HYD 20KTL-3PH, Firmware V113162

[image]

AdyRock, 2025-03-28:
Hmm, ok, I have obviously got the wrong register for your inverter. I will see if I can find out more.

blexter, 2025-03-28:
🙏

AdyRock, 2025-03-28:
OK, I have found a few likely candidates for the consumption register:
1157, 1159, 1160 or 1199.

So, could you open the app settings page and select the Register Log tab. The enter 1157 into the First Register Address box and 3 in to the Number of registers box and then click on the Start button.

[image]

(You will probably need to close the view as it appears there is a bug that prevents the start button appearing again.)

Then enter 1199 into the First Register Address and 1 into the Number of Registers and click on Start.

Once you have the results can you send post them back here.

blexter, 2025-03-29:
1157 = 160 1158 = 0 1159 = 0 1199 = 73 1199 = 72 1199 = 73

AdyRock, 2025-03-29:
1157 and 1199 look promising as they would equate to 1600 W and 730 W respectively. So, based on the 160 W in the image you posted from your inverter software, I would say 1199 is the most likely. Would you agree that at the time of recording the value you were using about 730W of power?

blexter, 2025-03-29:
Might be correct. 

[image]

AdyRock, 2025-03-29:
OK, I will go with that and post an update in a few minutes.

AdyRock, 2025-03-29:
I have posted the update.

AdyRock, 2025-03-29:
If it works OK, maybe you could try the Power Flow dashboard widget and provide some feedback.

blexter, 2025-03-29:
[image]

blexter, 2025-03-29:
The battery still seems to report 0% occasionally 

[image]

AdyRock, 2025-03-29:
Hmm, it seems there might be some other incorrect registers. 
If your panels are producing 6600W and you are only using 510W, then either the grid or the battery must be higher. 
Can you let me know what the values should have been for each of those at the time?

blexter, 2025-03-29:
Screenshots from same time, just a minute ago:

[image]
[image]

blexter, 2025-03-29:
2-3min later 
[image]

AdyRock, 2025-03-29:
Could you report registers 1157, 1160, 1284, plus the expected grid power.

It also seems the inverter has separate registers for each battery rather than a total. Do you have multiple batteries?

For the solar panels, could you check register 1198 as it seems the register I'm using is just for string 1.

I guess this will teach me not to use maps from HA 😁

blexter, 2025-03-29:
1 BMS, 10x2,5kWh batteries

[image]

1157 = 160 1158 = 0 1159 = 0 1199 = 73 1199 = 72 1199 = 73 1157 = Attempt to access memory outside buffer bounds 1160 = 296 1284 = 40 1198 = Attempt to access memory outside buffer bounds

AdyRock, 2025-03-29:
Sorry, I'm struggling to match the registers for the grid power. The document I have found is https://github.com/AdyRock/ady.sofar/blob/NewEnergySystem/Docs/SofarHYD%20Modbus%20protocol%20explaination-3ph.xlsx
Could you try the registered specified in that to see if you can track it down?
The number in the Address column is in hex so you can enter that in the Register Log tab by putting 0x in front of the number, for example 0x4AE for the "ActivePower_PV_Ext" field. The value returned is in the units specified in the Units column to the precision in the Precision column, for example, if 73 is returned with a precision of 0.01 then it would be 0,73kW.

[image]

If you can track down which registers provide the correct values I will update my map.

blexter, 2025-03-30:
Check all approx 3000 values? Or only the one in your picture? Or all with unit kW?
Then, which value should I compare with in this picture?

https://private-user-images.githubusercontent.com/37409960/428286847-b7bbf106-83fe-4b0c-baeb-0ac54de2708c.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDMzMTUwNjEsIm5iZiI6MTc0MzMxNDc2MSwicGF0aCI6Ii8zNzQwOTk2MC80MjgyODY4NDctYjdiYmYxMDYtODNmZS00YjBjLWJhZWItMGFjNTRkZTI3MDhjLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMzMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDMzMFQwNjA2MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05OTViNjkyYmI5ZTYyMDkxNjlhYjJjOWY1MmJhNzgzNDMxOGJlMDU5MTIzZWU4Y2MzNjU5ZmU3ZTkxMWMyMTllJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.r94oKuBKqoM1yMeBYaoQJDgKJCDrF8bhOapsvf5WYqU

AdyRock, 2025-03-30:
I can't see the picture, but you are trying to find the register that reports the grid power. So, check all the registers that mention power, probably active power, and compare the result with the reported power for the grid. I would suggest looking at the panel on the inverter (top right corner) as the solar app might not update very frequently. My best guess is register 1157 or 1160.

blexter, 2025-03-31:
Ok, i will check that out!
Just a headsup, my app/devices/connections have been frozen since two days now. Any troubleshooting in interest for you to do?

[image]

AdyRock, 2025-03-31:
Could you send the Diagnostics Log.

blexter, 2025-03-31:
> Could you send the Diagnostics Log.

Just checked "log data" and then send log, thats what i should do?

AdyRock, 2025-03-31:
Yes, it looks like some of the required data is missing quite frequently, but going by the time stamps, it should be updated in between the bad readings. 
Maybe enable the log so it captures more than just errors to make sure.

blexter, 2025-03-31:
Enable the log by check the box? ("Log Data"), and wait for a day or two and then send it again to you?
(I will try to get "fresh" data again by restarting the app and so on as well)

AdyRock, 2025-03-31:
Just enable the log for a few minutes and then send it again. 
My inverter will occasionally miss-report the data, so it seems to be a common issue. Hence, I added the checks. My suspicion is that it happens when the inverter writes to the register at the same time as the app tries to read it. Unfortunately, it doesn't always report an error, but just sends 0.

blexter, 2025-03-31:
Ok. just sent again. But the capabilities still updates (not frozen yet) so i might send it again after i see it been frozen?

blexter, 2025-04-01:
I just saw the values stopped update/been frozen for all sofar devices (4) since about 11h. I just sent you a logfile from the app.

blexter, 2025-04-22:
Sorry for delay! Im been busy and then on vacation.

1160 - grid in picture (right upper corner)
1167 - property usage (right bottom corner)
https://github.com/AdyRock/ady.sofar/issues/3#issuecomment-2763238997

This is what I think, due to delays in "running" the commands/recieve the result I'm not 100% sure.
Do you have all info, or should I check even more "Register addresses"?