AdyRock/ady.sofar #4: Problem to find the inverter

CYBERchata, 2024-09-04:
Hello Adrian,
I have for few days Sofar HYD 6KTL-3PH inverter with LAN logger.
I can't add this device to your App, because nothings found.
Is this a problem with this device support or something else?
If you can add the support for this model, I will be grateful.
Maybe problem is something else, what can I do to make it work in Homey?

AdyRock, 2024-09-09:
Have you tried the test version of the app?

AdyRock, 2024-09-09:
The test version has just gone live, so try the updated version.
If that doesn't work, then send the log from the app settings page.

CYBERchata, 2024-09-09:
Nothing's found, inverter found but no results. Log was sent directly from the app.
Thanx for help..

AdyRock, 2024-09-10:
Your inverter hasn't reported anything from the grid frequency register of the 4 known registers. So, basically the app doesn't have a suitable register map for your inverter. I did a quick search and I can't find a description of the registers for the model you have. I will try again this evening when I get home from work, but if you have a link to the location of a firmware update please let me know.

CYBERchata, 2024-09-10:
I have only hybrid inverter and battery.
The firmware and other things I get from here:
https://downloads.sofarsolar.eu/?_product=hyd-5-20ktl-3ph

and modbus registers paste here:

[SOFAR_Modbus_Protocol_English_G3_V1.22(only 3PH).xlsx](https://github.com/user-attachments/files/16940076/SOFAR_Modbus_Protocol_English_G3_V1.22.only.3PH.xlsx)
[SOFAR-G3 External Modbus Protocol-EN-V1.10 (20220622).xlsx](https://github.com/user-attachments/files/16940077/SOFAR-G3.External.Modbus.Protocol-EN-V1.10.20220622.xlsx)

Thanx a lot.

AdyRock, 2024-09-10:
That is the table I was looking at before which should mean your inverter provides the Grid Frequency in register 1156 and that is supported. However you inverter seems to return NULL:
```
Checking register 1156 for grid frequency:

* 2024-09-09T22:15:01.860Z
send_request: start = 1156, end = 1156, fc = 3 

* 2024-09-09T22:15:02.975Z
Returned null.
```
so, it seems that is not the correct file for it.

CYBERchata, 2024-09-10:
It's strange, this registry table is in firmware zip to this inverter model. Today I've upgraded firmware in inverter but nothing's change. I have a connection via cable lan logger, not via WiFi but it doesn't matter, right?

AdyRock, 2024-09-10:
I have no idea if that makes any difference.
It looks good as it is advertising and detected, but not responding.
In the app settings page, there is a tab to try different registers, so maybe have a play in there to see if you can get any response.

CYBERchata, 2024-09-10:
I'll try to contact with local Sofar distributor, maybe they can help and explain this problem. Thanx for help, I'll back with response if I will know anyrhing more. Thanx.

closed 2025-03-27 by AdyRock