AdyRock/ady.sofar #5: Reversed signs in energy tab

imydal, 2025-04-06:
Homey pro's energy tab shows energy storage and consumption from the grid with reversed signs. They turn out negative instead of positive.

Production from the panels is shown correctly.

[image]
[image]

I attach screenshot 
Thank you very much

AdyRock, 2025-04-06:
Can you make sure you are using the test version as that has been optimised for the new energy tab.

imydal, 2025-04-06:
Is the 1.1.24 version

[image]

AdyRock, 2025-04-06:
The test version is 1.1.28. To get that, find it in the store and add test to the end of the url, then install it.